uf-mil/NaviGator #24: Turn Shooter into ROS Service

zachgoins, 2016-08-06:
We need a way to be able to fire the launcher from other programs using something like

shooter.fire(), shooter.cancel(), etc...

Basically just opens up the commands you would manually input into the terminal to the ROS server system where we can call the functions from any programs. 

http://wiki.ros.org/rospy/Overview/Services

LucasBA, 2016-08-06:
Should this be assigned to @ironmig ?

rleyva, 2016-08-06:
I added him to uf-mil. Once he accepts the invite we'll assign him to the issue.

ironmig, 2016-08-06:
Any suggestions on how to lay this out? Is it better to have a unique service for each operation (shoot, cancel) or just one service where you pass data about what operation to do?

rleyva, 2016-08-06:
Take a look at sub8_actuator_driver, the service model there should port over directly. Here is the [link](https://github.com/uf-mil/Sub8/blob/5feeed1af81630d056c772f909712621ea6f0c77/drivers/sub8_actuator_driver/srv/SetValve.srv) to the service. Take a look at the readme in the package for more info.

zachgoins, 2016-08-07:
The link that Ralph gave is great, but yes typically there is a different
service for each action. They can all be in the same server file but we
want to be able to call your command line inputs directly ie
shooter.shoot(), shooter.cancel(), and then the server interacts with the
shooter script appropriately.

*Zach Goins | 918.801.8242*
paracosm.io
Github <https://github.com/zachgoins> | LinkedIn
<https://www.linkedin.com/in/zachgoins>

closed 2016-09-22 by ironmig