uf-mil/NaviGator #29: Continuous Integration

mattlangford, 2016-08-17:
Using Semaphore Continuous Integration requires admin privileges of the repo to set it up. @zachgoins  are you the admin? Since we have the install script now, it should be simple to actually set it up but you need to be the git admin.

mattlangford, 2016-08-18:
Looks like it's working now. @DSsoto is working to cache the ros install so that we don't have to redownload all the ros packages every time we want to do a PR.

mattlangford, 2016-08-19:
To a bigger linux guru then myself, this might be simple. There's a `.semaphore-cache` that the ros install files could be downloaded to and then at run time installed from. For info check [this](https://semaphoreci.com/docs/caching-between-builds.html) out.

sub8, 2016-08-19:
You may want to look into using Docker on Semaphore, though that may open
another can of worms

khaledhassan, 2016-08-19:
Hmm. Not sure why my email reply above didn't get attached to my name

jpanikulam, 2016-08-19:
You heard it here first, folks. Khaled is the sub. 

- Jake

mattlangford, 2016-08-19:
I was really confused as to who that was, especially since the face kinda looks like the sub

khaledhassan, 2016-08-19:
***tl;dr: I actually am the sub***

I figured out why. Once upon a time I created a dummy github account for the sub to attach the ssh key (so that minor changes could be made and pushed from the sub itself?). The email address I used happens to forward to my main inbox, which also explains why I get two copies of every email. I evidently replied to the sub8 version of the email, not the khaledhassan version.

closed 2016-08-27 by mattlangford