uf-mil/NaviGator #39: Semaphore Build Failing

mattlangford, 2016-08-26:
It fails for almost every build. No idea what the problem could be. See me if youd like ssh access into the semaphore VM.

We could make our own CI system that wouldn't be garbage by having a script on mil plumbusi that pulls from any pr branches it sees and tests the build on there. Not sure how we would get the cool checkmark or x that semaphore has tho

whispercoros, 2016-08-27:
I'll look in to getting this set up on plumbusi with some pre-built CI solution if it's not resolved soon. I think going through all the trouble to make our own would kill a lot of time, so let's hope one is available.

mattlangford, 2016-08-28:
As per the recommendation of @ironmig, I added a `catkin_make clean` before the final compilation, and everything seems to work fine now.

closed 2016-08-28 by mattlangford