uf-mil/NaviGator #42: Build Issue

LucasBA, 2016-08-31:
When running catkin_make on the current master branch I got this error:
--------------------------------------------------------------------------------------------------------------------------------------------
In file included from /home/user/nav_ws/devel/include/navigator_msgs/ShooterManual.h:8:0,
                 from /home/user/nav_ws/src/NaviGator/mission_systems/shooter/firmware/shooter.cpp:5:
/opt/ros/indigo/include/ros/service_traits.h:31:50: fatal error: boost/type_traits/remove_reference.hpp: No such file or directory
 #include <boost/type_traits/remove_reference.hpp>
                                                  ^
compilation terminated.
make[6]: *** [CMakeFiles/arduino.dir/shooter.cpp.obj] Error 1
make[5]: *** [CMakeFiles/arduino.dir/all] Error 2
make[4]: *** [CMakeFiles/arduino.dir/rule] Error 2
make[3]: *** [arduino] Error 2
make[2]: *** [NaviGator/mission_systems/shooter/CMakeFiles/navigator_shooter_firmware_arduino] Error 2
make[1]: *** [NaviGator/mission_systems/shooter/CMakeFiles/navigator_shooter_firmware_arduino.dir/all] Error 2
make: *** [all] Error 2
Invoking "make -j4 -l4" failed
 @--------------------------------------------------------------------------------------------------------------------------------------------
Matt Langford said he and Tess ran catkin_make clean to fix

mattlangford, 2016-08-31:
Did you get it working when you cleaned it?

LucasBA, 2016-08-31:
kinda. Output of clean:
--------------------------------------------------------------------------------------------------------------------
$ catkin_make clean
Base path: /home/lucas/nav_ws
Source space: /home/lucas/nav_ws/src
Build space: /home/lucas/nav_ws/build
Devel space: /home/lucas/nav_ws/devel
Install space: /home/lucas/nav_ws/install
####
#### Running command: "make cmake_check_build_system" in "/home/lucas/nav_ws/build"
####
####
#### Running command: "make clean -j4 -l4" in "/home/lucas/nav_ws/build"
####
--------------------------------------------------------------------------------------------------------------------------

If I run catkin_make again without clean I get the error again

whispercoros, 2016-09-02:
As much of a hack as this sounds like, it worked for me with the build issues:
1. Completely delete the build/ folder in mil_ws/
2. Run catkin_make -B
3. Run catkin_make to ensure that it will work after the first build

Steps 1-2 can be thought of as a "forced clean build" of sorts, so I find it strange that this works for me. If it works on your machine, let me know.

ironmig, 2016-09-07:
I've been wresting with this error since when I began working with ros arduino. Basically, the shooter CMAKE creates a folder of all the message/service/ros headers to be compiled for arduino hardware. But if something goes wrong in generating the messages or if the shooter gets compiled before some messages, weird stuff happens. Doing what Anthony suggested does work. This works too, and is a little faster:

1. ```rosdep update ```

1. ```rm -R build/Navigator/mission_systems/shooter/ros_lib```

2. ```catkin_make navigator_msgs_generate_messages```

3. ```catkin_make navigator_shooter_firmware_arduino```

It seems that once you get it to build correctly once, you're good. IDK... I could just get rid of the custom service and just use std::float or something to prevent this error.

LucasBA, 2016-09-07:
I tried Kevin's fix first (because I didn't see yours @whispercoros ) and works for me. Should it be added to the end of the install script?

mattlangford, 2016-09-07:
Do you think it's generating messages in the wrong order? Like it tries to move the messages before they are generated

LucasBA, 2016-09-09:
Thats what it seems like

LucasBA, 2016-09-09:
The shuttle is having this issue now:
/home/uf-mil/mil_ws/src/rawgps-tools/rift_dk2_imu_driver/src/rift_dk2_imu_driver_nodelet.cpp: In constructor ‘rift_dk2_imu_driver::NodeImpl::NodeImpl(boost::function<const std::basic_string<char>&()>, ros::NodeHandle*, ros::NodeHandle*)’:
/home/uf-mil/mil_ws/src/rawgps-tools/rift_dk2_imu_driver/src/rift_dk2_imu_driver_nodelet.cpp:178:26: error: ‘in_place’ is not a member of ‘boost’
         tracker_thread = boost::in_place([this]() {
                          ^
/home/uf-mil/mil_ws/src/rawgps-tools/rift_dk2_imu_driver/src/rift_dk2_imu_driver_nodelet.cpp: In member function ‘virtual void rift_dk2_imu_driver::rift_dk2_imu_driver_nodelet::onInit()’:
/home/uf-mil/mil_ws/src/rawgps-tools/rift_dk2_imu_driver/src/rift_dk2_imu_driver_nodelet.cpp:219:20: error: ‘in_place’ is not a member of ‘boost’
         nodeimpl = boost::in_place(boost::bind(&rift_dk2_imu_driver_nodelet::getName, this),
                    ^
make[2]: *** [rawgps-tools/rift_dk2_imu_driver/CMakeFiles/rift_dk2_imu_driver_nodelet.dir/src/rift_dk2_imu_driver_nodelet.cpp.o] Error 1
make[1]: *** [rawgps-tools/rift_dk2_imu_driver/CMakeFiles/rift_dk2_imu_driver_nodelet.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
Linking CXX shared library /home/uf-mil/mil_ws/devel/lib/libsub8_vision_lib.so
Linking CXX shared library /home/uf-mil/mil_ws/devel/lib/libmagnetic_dynamic_compensation_nodelet.so
Linking CXX shared library /home/uf-mil/mil_ws/devel/lib/libgazebo_ros_paths_plugin.so
[ 72%] Built target magnetic_dynamic_compensation_nodelet
[ 72%] Built target sub8_vision_lib
[ 72%] Built target gazebo_ros_paths_plugin
Linking CXX shared library /home/uf-mil/mil_ws/devel/lib/libnavigator_vision_lib.so
[ 72%] Built target navigator_vision_lib
Linking CXX executable /home/uf-mil/mil_ws/devel/lib/sub8_controller/pd_controller
[ 72%] Built target pd_controller
Linking CXX shared library /home/uf-mil/mil_ws/devel/lib/libgazebo_ros_api_plugin.so
[ 72%] Built target gazebo_ros_api_plugin
make: *** [all] Error 2
Invoking "make -j8 -l8" failed

I tried both anthony and kevin's fixs. Ill keep trouble shooting later but Im going home so I figured I would post this.

whispercoros, 2016-09-11:
I received the same error when I attempted to use the shuttle. It appears that the issue is with the rawgps repository, but that's not really possible as it has been working for months. Worst case scenario, I can just reinstall everything on Monday.

mattlangford, 2016-09-11:
I've switched over to using the old computer for simulations - I think I software cleanse is warranted on that thing

forrestv, 2016-09-11:
Hey, I think I may have fixed the rawgps-tools problem with commit 39fc8aad6 in that repository. I think it was just a missing #include as per http://stackoverflow.com/questions/32047745/missing-boostin-place-in-1-59.

whispercoros, 2016-09-12:
I will be waiting for Ralph to back up his files on the shuttle before reinstalling it. If anyone else has important files on the machine, I recommend getting them backed up soon.

whispercoros, 2016-09-23:
@mattlangford, can you take a look at Semaphore? I was able to get build working again after #57 was reverted. It fails on the following line which I believe you added:

    python -c "import nav_missions"

whispercoros, 2016-09-26:
I commented out the line in Semaphore so we have an actual build status for now

closed 2016-10-12 by whispercoros