uf-mil/NaviGator #43: Simulator issue

viron11111, 2016-09-01:
When the simulator starts up, the challenges do not appear.  We are missing all of the challenges except for the.  Also, I cannot see land in the simulator.

mattlangford, 2016-09-01:
I am also having this issue

whispercoros, 2016-09-02:
I have been working on the simulator over the past few days and am planning on bundling everything I have together in a PR sometime this weekend.

mattlangford, 2016-09-02:
Just be sure to use gitlfs for any models

RustyBamboo, 2016-09-02:
Last time I loaded it there were buoys... Now I can't get them to load.

tessbianchi, 2016-09-02:
Yeah that's a really good idea!

ironmig, 2016-09-02:
It would be very helpful to have some dock signs up there with different colors and shapes

viron11111, 2016-09-07:
Did you bundle everything into a PR this weekend?

LucasBA, 2016-09-09:
> 

> ``

LucasBA, 2016-09-09:
It seems like the boat model is breaking the sim. getting [gazebo-18] process has died

mattlangford, 2016-09-09:
Were you able to run it without the boat model?

LucasBA, 2016-09-09:
Yes

mattlangford, 2016-09-09:
Can you try to open the model in blender?

LucasBA, 2016-09-09:
Ill try when I get back to the lab

whispercoros, 2016-09-11:
The install script will automatically default to mil_ws/ now, but you can keep navigator_ws/ if you want by passing it with this:

./install.sh -c navigator_ws/

In #49, I added git-lfs installation and pull to the install script, so this should take care of any of those issues.

whispercoros, 2016-09-11:
I have been working on a system that automatically generates the courses based on the specifications provided in the competition rules. Since this is taking longer than expected, I am just going to make a sandbox with all course elements in it by hand for the time being.

tessbianchi, 2016-09-13:
Make sure that you run roslaunch navigator_launch simulation.launch, and don't directly run the goose.launch file. Should we mark as resolved?

mattlangford, 2016-09-13:
So make sure you've `git lfs pull` so hat you have the most recent models, then like Tess said do `roslaunch navigator_launch simulation.launch`

mattlangford, 2016-09-22:
We need challenges in the sim.

DSsoto, 2016-09-22:
Anthony, it would be really useful we could have the challenges in the simulator.  Please add the challenges to the simulator.

viron11111, 2016-09-22:
Don't worry guys.  Anthony told me he would have them finished by last night.  He's probably done with it and only needs to upload the updated files to Git.

whispercoros, 2016-09-23:
I have most of the challenges ready in the sim, but Git-LFS keeps giving me errors when I attempt to pull and push from my local copy the repository. I am doing some research, but I will be in MIL tomorrow to see if anyone else has had similar issues.

tessbianchi, 2016-09-23:
Anthony....where are my challenges?

ironmig, 2016-10-11:
@whispercoros Have there been any updates on this? Are we going to bring the simulator changes into the main repo with a PR soon?

whispercoros, 2016-10-12:
@ironmig Since I still need Lucas to model a few components, I will split this up into two PRs. I will merge your changes into the sandbox and open a PR to dev today.

viron11111, 2016-10-20:
@whispercoros have you received the model from Lucas yet?

whispercoros, 2016-10-21:
@viron11111, I received the models and have Detect and Deliver implemented in the simulator. It will be included in #83, which should be merged to dev on Saturday (10/22)

viron11111, 2016-10-21:
Roger.

whispercoros, 2016-10-31:
This has been fixed by #83

closed 2016-10-31 by whispercoros