uf-mil/NaviGator #45: Thrusters not giving feedback

mattlangford, 2016-09-06:
Not sure if it's a hardware, firmware, or software issue. The thrusters won't return feedback information (we mainly just want battery voltage)

whispercoros, 2016-09-11:
Any progress on this? It's not vital, but it would be nice to have a voltage indicator for testing.

mattlangford, 2016-09-19:
Thrusters only fail to give feedback when launched from `master_boat.launch`. When launched by themselves everything seems to be fine.

mattlangford, 2016-09-26:
The problem here is that if the motor drivers start up after the thruster mapper starts up, the motor driver seems to not ever poll for feedback from the thrusters. Solution for now is to sleep before the thruster mapper comes up so that the feedback polling will still continue. Fixing this would require some personal time with the roboteq thruster drivers.

PR #69 (2016-09-30): Updates from lake day
This should fix the "17 commits behind" thing in dev.

Also fixes #45 - albeit it's a patch at best.

PR 69 is best PR
closed 2016-09-26 by mattlangford