uf-mil/NaviGator #46: Judges Display (GUI)

viron11111, 2016-09-07:
We need a GUI for the judges.  See pages 27-28:
http://robotx.org/images/files/2016-MRC-Tasks-2016-07-13.pdf

whispercoros, 2016-09-11:
I would be able to get something together for this in QT in a few days, but right now it is not high on my priority list as we have no data to report to it right now. Matt has also expressed interest in implementing a web-based solution if he has some extra time. Either way, this will likely be one of the last things we finish.

mattlangford, 2016-09-11:
I think Ira had started work on this

viron11111, 2016-09-11:
Roger.  Yes, Ira has started working on this.  I forwarded it to Anthony.  It needs to be ROSified though.

And we do have data to report right now.  Tess can detect colors and Kevin/Daniel can detect shapes.

The GUI is a required by the rules.  I do not want to procrastinate and work on it at the last minute.

maverick7170, 2016-09-15:
Sorry for not responding earlier! The GUI is done with ROSification; just need to decide on exact method of communicating data (topics/services/etc..).

viron11111, 2016-09-15:
Great!  Tess will be the first one who can provide you with an actual response.  Yesterday, Tess stated that she was satisfied with the sign detection portion (at this time) and was starting work on the mission.  If she makes enough progress and Matt G can fix the sign's electronics, we may be able to do the mission  and include the GUI.  ROS params was mentioned at the last lake day...

whispercoros, 2016-10-18:
Ira said he would email me the GUI code sometime this week. Once I integrate it, I will close this.

viron11111, 2016-10-20:
Let's not close this until we have the GUI implemented and using messages from our missions.

viron11111, 2016-11-13:
Now that Tess's code works, can we test the GUI today?  Ira said that it should work.

whispercoros, 2016-11-13:
The GUI test was successful for Scan The Code

viron11111, 2016-11-14:
Roger.  Has it been pushed onto the master?

whispercoros, 2016-11-15:
I think it is still on Ira's fork, he will need to PR it when he feels it is complete.

closed 2016-11-22 by whispercoros