uf-mil/NaviGator #51: RobotTeq motor controller current trip

viron11111, 2016-09-12:
Need to update firmware on controller to trip at 70 amps (currently 60 amps).

mgriessler, 2016-09-12:
I have a feeling the controller will then just push the motors harder, and we’ll trip at 70A.

viron11111, 2016-09-12:
Roger.  I think the current spike is happening when the thrusters are rapidly switching directions.  I'm going to slow the ramping speed in the firmware.  This will increase the time it takes to switch directions but reduce the current spikes.

viron11111, 2016-09-13:
Jason is confident that his controller is not the problem.  I'm still looking into the firmware.  From Jason:

"That is not the case.

    The thruster mapper hard-limits its outputs. They never exceed the safe maximum.
    Both controllers I have run are continuous. They are substantially smoother in what they demand than any human could request with the xbox-controller.
    The xbox-controller maps full analog-stick tilt to maximum thrust. That means when you are driving it around, you are slamming the thrusters with max-thrust requests. (This is not a problem since the maximum is limited to something safe, but the autonomous controllers are always kinder than this).
    I watch the desired wrench on the screen at all time- it doesn't oscillate rapidly because the controller is PD + an integral. The PD term is linear in the state (which changes slowly with respect to thruster speeds) and an integral which is necessarily smooth. There are no sliding-mode or high frequency terms in any controller I have ever run on the boat.
    These controllers both have had substantial test with no problems. In fact, MRAC has been running since day 1, and CL has about 2 hours of straight run time at this point."

viron11111, 2016-09-15:
Gave new .hex file to Matt for implementation.  Implements a 25% reduction in acceleration and deceleration.  Hoping this will reduce current spikes.

viron11111, 2016-09-17:
We're holding off on implementing new .hex file.  Jason wants to keep the ramping rates the same for now.  Still not sure why the motor controllers needed to be reset...

viron11111, 2016-09-20:
Still not sure why the motors need to be reset.  Work around, detect when the motors stop responding, reset motorcontrollers with Daniel's kill board.

ironmig, 2016-10-12:
Any progress on this? We should start closing issues if they've been resolved.

mattlangford, 2016-10-14:
Hasn't been resolved but hasnt been observed again whilst testing

mattlangford, 2016-10-19:
Bump
We had a thruster drop out this past weekend. There was no feedback coming from any of the thrusters so I couldn't check that.

viron11111, 2016-11-14:
I'd like to consider this partially solved by switching out batteries mid-day.  Agree?

whispercoros, 2016-11-15:
@viron11111, The current trips were very rare even without the battery swapping, so I do not think enough testing has been done to verify this. It would be nice to have a way to reset the motor controllers in software, but I do not think it is a high priority objective at the moment.

viron11111, 2016-11-15:
Roger.

mattlangford, 2016-11-26:
I think this happened both of our last two test days. I'd say this has gone up as a concern in my mind.

viron11111, 2016-11-26:
Did it happen early in the day or late?  Testing anything unusual?

mattlangford, 2016-11-26:
The battery voltages were around 24 when it failed. This is going to be problematic if we remove the old kill switch since the new kill switches directly cut off power to the motors and don't restart the motor drivers (the only way we know how to fix them). @mgriessler took a look when we lost power and he said that there was nothing out of the ordinary with the motor drivers (no error LEDs or obvious problems).

The whole inverter dropped out at least once the last weekend as well, and I think this was the cause of most of our network issues in the past. We had a powered USB hub (with the INS and all four FTDI USB motor cables plugged into it) and the POE plugged into the inverter and when network came back up, odom dropped and the motors were unresponsive. After I shuffled USB devices around and restarted the computer things worked again.

I think we should try to remove things from the inverter as much as possible since it seems to drop out. Since most of the devices take AC from the inverter and convert it to DC with a wall wart, could we skip the AC conversion and power it directly from one of the power busses we have onboard? I think it's definitely worth putting some effort into since network loss would be devistating during a practice round and embarrassing if we had to row the boat back to shore.

viron11111, 2016-11-26:
Roger.  POE takes a weird voltage so it won't be easy to replaced the hardware.  Working with Matt G.  What happened to switching out the batteries halfway through testing?

whispercoros, 2016-11-26:
I spoke to Daniel Dugger about this on the last test day. Both the PoE and switch could be replaced with the same switch that was bought for shore operations. It would also be fairly easy to get a hold of a second PCI USB bus so that the need for a powered hub could be eliminated. This would allow us to remove the inverter all together and make room to properly mount his kill board to the chassis. This not only cleans up the wiring and makes the system more robust, but eliminates the unnecessary power consumption of the inverter.

mattlangford, 2016-11-27:
We hadn't been able to fully charge the replacement set if I remember correctly. So the whole day we operated in the low zone.

whispercoros, 2016-12-13:
The solution I detailed above has been implemented with some minor alterations. We are free from the tyranny of the inverter!

closed 2016-12-13 by whispercoros