uf-mil/NaviGator #52: GPS waypoints

viron11111, 2016-09-12:
NaviGator is having problems going to GPS way points.  Boat local Lat/long are very far off from Google Earth points.

Correct me if I'm wrong but it seemed that the boat was unable to return to recorded (using Sylphase INS) points using own INS system.  Maybe an error from translation of relative to world?

forrestv, 2016-09-12:
Looking at the code, the (first) problem I see is that `PoseEditor2.latlon` calls `self.rel_position(enu_vector)` at the end, but `rel_position` expects a *body-frame* vector, not a world (ENU)-frame vector.

@mattlangford, can you confirm?

Also, @mattlangford, can you explain what's preventing us from catching errors like this in simulation?

mattlangford, 2016-09-12:
I agree, it appeared to be driving between two waypoints, but that could have been a quark where the frames had lined up - should be an easy fix.
The reason it's not right now is just that no one has worked on it. I think it's important, so I will spend some time in the near future setting up a more complete navigation stack in sim.

closed 2016-09-26 by mattlangford