uf-mil/NaviGator #55: Racquetball double feed

viron11111, 2016-09-13:
This weekend, we double fed the ball launcher during the last run.  This was caused by starting a mission and killing it before the mission completed (ball loaded but not fired).  When we re-attempted the mission, the mechanism fed a 2nd ball into the chamber.  Daniel thinks that this can be solved by using ROS action library.

ironmig, 2016-09-17:
Should be fixed in #60. Going with a simple check rather than action library for now. If you call the load service and get success == false, it is either already loaded or currently doing something else (loading/firing). Likewise, if you call the fire service and success == false it is either not loaded or currently doing something.

ironmig, 2016-10-04:
Shooter uses actionlib once #71 is merged.

closed 2016-10-07 by ironmig