uf-mil/NaviGator #89: Side camera (for shooter) has unreliable auto whitebalance/exposure/brightness

ironmig, 2016-10-16:
The pointgrey camera on the right side of navigator, under the shooter, is unreliable in is auto configuration as compared to the front cameras, which look quite good at any of the tested conditions. This could be for a few reasons:

- The camera is physically in a tube which blocks a significant portion of its field of view the border of which is white. This could throw off the auto white balance or other settings.
- The settings on the camera/driver need to be tuned more

ironmig, 2016-10-16:
Note: this problem greatly affects the reliability of the detect deliver symbol perception

ironmig, 2016-10-16:
Another idea discussed is the set the ROI of the camera driver to ignore the borders of the waterproofing tube. Will this change what data is used for auto white balance?

mattlangford, 2016-10-19:
Hopefully the new lens will fix this - manually setting certain camera settings doesn't really solve anything (as we saw) since looking at objects from different angles changes the light in the scene drastically.

viron11111, 2016-10-20:
I still have the old lens if we need to go back.

ironmig, 2016-11-10:
Seems to be significantly improved with new camera + lens. Will be even better with downward angled mount.

closed 2016-11-10 by ironmig