uf-mil/NaviGator #90: Object classifcation database should return the original data (Buoy.msg)

ironmig, 2016-10-16:
Currently, a service call to /database/single or /database/full returns data in the format of PerceptionObject.msg, which contains the objects id, name, size, and position. This removes some of the data of the original Buoy.msg sent by object detection, such as a normal and the 3D points of the object. For use in missions and for less confusion, perhaps this should just be one message format?

mattlangford, 2016-10-17:
I think the plan right now is for @maverick7170 (Ira) to get a database full of automatically classified objects for this weekend. I'm going to assign him on this too - make sure he knows if you want more information in the message then what currently exists.

maverick7170, 2016-10-19:
Yes, I'll have this ready for this weekend!

closed 2016-11-01 by ironmig