uf-mil/NaviGator #112: Simulator should not launch tf.launch

ironmig, 2016-11-01:
One thing holding me back from making serious use out of the simulator has been apparent differences between the location of sensors on Navigator in gazebo and the transformations publishes by tf.launch. Rather then trying to get these numbers just right, perhaps the simulator should just publish it's own TF's based on the boat model. A quick Google seems to show that this may even be the default behavior of a ros-ified gazebo.

ironmig, 2016-11-01:
http://wiki.ros.org/gazebo2rviz

whispercoros, 2016-11-01:
I was under the assumption that the locations of the simulated sensors as well as the TFs for the whole system were changed when Lucas measured everything. Was this not the case? Regardless, I think it is a worthwhile investment to have the sensors line up as close as possible to real life in the simulator, so I will look into this during the week.

viron11111, 2016-11-01:
Anthony,

It's been almost a week.  Have you found a button (from Amazon) for the 
2nd computer yet?

-Andy

ironmig, 2016-11-04:
From looking at how Sub8 did it, the problem is that our tf.launch specifies sensor locations in reference to measurement and gazebo is with reference to base_frame. I wrote a script to generate the correct values and will submit a PR with the fixes today after testing.

whispercoros, 2016-11-04:
I am interested to see if this works, have you tested it? I believe the model of our boat has it's base_frame point at measurement. You should be able to check this in the simulator fairly easily.

ironmig, 2016-11-05:
Tried to figure this out, could not. This is pretty high priority, as nothing really works without good TF's.

whispercoros, 2016-11-10:
@ironmig When you were troubleshooting, was the issue with the position or rotation of the sensors?

ironmig, 2016-11-10:
Hard to tell. Getting the rotation to be the same as the tf.launch is tough because cameras have their axises switched in tf but use the normal axises in gazebo. Might also have something to do with camera projection. A good first step is to finish that program Matt and I were working on to convert tf to sdf <pose> tags.

whispercoros, 2016-11-11:
Is this program on upstream or one of the workstations?

ironmig, 2016-11-11:
Fixed in #127

closed 2016-11-11 by ironmig