uf-mil/NaviGator #115: Simulator identify & dock model is inaccurate

ironmig, 2016-11-03:
Considering we will have almost no testing with the identify and dock challenge before hawaii, we really need a good model for the sim. Right now there is no back panel, and the displayed symbols are too dark (like detect deliver used to be). The model needs to be fixed and make it easy to change which symbols.

whispercoros, 2016-11-04:
I believe Lucas has completed replacement models; I will put those in the simulator once I edit the coloring on them.

whispercoros, 2016-11-06:
The models have been updated on my fork and will be in my next Simulator PR. I will also be making the dock pieces a solid object so that it is easier to move around. The panels can all be swapped out by editing the launch file.

whispercoros, 2016-11-11:
This is fixed in #128.

closed 2016-11-11 by whispercoros