uf-mil/NaviGator #116: Simulator does not publish bounds

ironmig, 2016-11-03:
The simulator does not launch a bounds server (it can't because it doesn't have absolute odom), so the ogrid, and therefore everything that relies on it (obstacle avoidance, missions) do not work in the sim. This can be a pretty quick fix , jut publish bounds that are in a square the size of the gazebo world.

mattlangford, 2016-11-03:
Fixing in next PR

whispercoros, 2016-11-04:
I would prefer just turning off the bounds in the simulation, I believe @maverick7170's OGrid is the only thing that does not listen to the enforce_bounds parameter.

ironmig, 2016-11-04:
We need some bounds, as it determines the size (width by height) of the ogrid to publish. Doesn't seem like a big deal for the bounds server to just check /is_simulatiom and put some fake ones in.

mattlangford, 2016-11-04:
I'm putting real GPS coordinates into the sim, so everything will work just like normal

ironmig, 2016-11-10:
Added in #125

closed 2016-11-10 by ironmig