uf-mil/NaviGator #121: Simulator launching problems

ironmig, 2016-11-04:
There seems to still be some frustrating problems trying to launch the sim.  It is probably a matter of timing. Sometimes gazebo will just freeze on a black screen, sometimes gazebo doesn't come up at all. Sometimes it freezes in spawning models. Happens on slow computers and fast computers

Don't really have a more detailed bug report. Just launch it a couple times in a row and you’ll see some issues.

mattlangford, 2016-11-04:
Try echoing some of the topics (`/odom`, `/velodyne_points`, `/stereo/left/image_raw`) to see if the backend is coming up and it's just the front end crashing. We've had this problem with the sub too, I guess it could be something not starting up at the right time.

ironmig, 2016-11-05:
Correct, it is mostly a problem with just the front end when using sandbox mode. Perhaps there needs to be a delay / wait to allow gazebo to load before loading models. There are also various other synchronization issues like launching obstacle detection before bounds have been set.

ironmig, 2016-11-10:
Maybe simulator.launch just shouldn't launch the gazebo client. Add in the readme somewhere how to launch the client separately (once simulator.launch is done)

whispercoros, 2016-11-10:
I have been working on the simulator sporadically over the past week and the only time I have ever experienced a crash on launch is when the models from Git LFS were not up to date. I should point out that I only ever use shuttle or johnny-five for testing, though.

whispercoros, 2016-11-10:
We have had a ton of issues attempting to sever the frontend from the backend in the past. Though it's supposed to be possible, and was even the plan for shuttle and johnny-five (headless gazebo servers that we connect to with the GUI), nobody was able to get it working.

whispercoros, 2016-11-10:
Is there a specific out-of-order launching issue you have experienced, or do these happen at random? If it is a specific issue, we could move some things around in the launch file to try to get the ordering right.

ironmig, 2016-11-11:
I think my issues were mostly from gzclient, which can be not run with the ``gui:=false``` argument.

closed 2016-11-11 by ironmig