uf-mil/NaviGator #126: RobotX write up

viron11111, 2016-11-09:
Missing write up on simulator, acoustic based transit, and stereo vision.

whispercoros, 2016-11-10:
The simulator write up has been submitted.

viron11111, 2016-11-10:
Roger.  Stereo vision, acoustic based transit, and the state machine write ups are the only missing portions from the software team.

tessbianchi, 2016-11-10:
Who is doing the state machine?

viron11111, 2016-11-10:
I wanted Matt to do it but he said you may be working on it already.  I assumed that we had a section for missions but it's not on the Google doc.  Tess, can you knock it out?  Just a quarter page on how our missions transition (txros stuff) with time outs ect...

viron11111, 2016-11-10:
Have received write ups on stereo and acoustic based transit.  Last is the state machine.

whispercoros, 2016-11-13:
@viron11111, can this be closed now that the paper deadline has passed?

viron11111, 2016-11-13:
Still waiting on machine.  Tess said she would complete it on Tuesday.  Cutting it close!

viron11111, 2016-11-14:
90% written as of this morning.  Missing electrical section, state machine write up, CAD model.

closed 2016-11-20 by tessbianchi