uf-mil/NaviGator #132: Write coral survey perception and mission

ironmig, 2016-11-14:
So far, no work has been done to write the perception or mission for the Underwater Shape Identification (previously Coral Survey) challenge. By the competition, we should have the mission written and the best perception we can get based on the sim, [the footage in hawaii](https://www.youtube.com/watch?v=JDpB1n8pPfs), and potential testing in a pool.

whispercoros, 2016-11-15:
@DSsoto is working on the perception this week and I will be writing the actual mission seeing as not much progress has been made here. We hope to have a preliminary version tested in the simulator before Saturday (11/19/16).

tessbianchi, 2016-11-15:
Once again, please do not split missions.

DSsoto, 2016-11-15:
@tessbianchi, I disagree. You could argue that ultimately there should be only one person responsible for a mission  once we get to Hawaii, but in the meantime, teamwork and parallel development lets us get stuff done more quickly.

ironmig, 2016-12-06:
#172 adds some work towards perception, though right now it does not have an interface to ros. @whispercoros what is the status of the mission code?

closed 2016-12-18 by mattlangford