uf-mil/NaviGator #133: Wrte find the break perception and mission

ironmig, 2016-11-14:
So far, no work has been done to write the perception or mission for the find the break challenge. By the competition, we should have the mission written and the best perception we can get based on the sim, [the footage in hawaii](https://www.youtube.com/watch?v=JDpB1n8pPfs), and potential testing in a pool.

whispercoros, 2016-11-15:
@DSsoto is working on the perception this week and I will be writing the actual mission seeing as not much progress has been made here. We hope to have a preliminary version tested in the simulator before Saturday (11/19/16).

tessbianchi, 2016-11-15:
I've talked with @DSsoto about this previously. 
Splitting missions is not a good idea. It diffuses responsibility so that no one person takes full ownership of the mission and creates over head in debugging because two people will have to debug one mission instead of one person.

We will all be very busy in Hawaii, and having two people working on something that one person can do is a waste of valuable manpower. 

Also, txROS takes 1-2 days to get the hang of MAX.

whispercoros, 2016-11-15:
I agree that this is how we should operate; however, I'd like to bring up a few things to try to sway you on this:
* The same perception tool will be used on both Coral Survey and Find the Break, meaning that if it is one person to a mission, one person will have to undertake both missions alone
* @DSsoto has a very strong background in perception (which I do not) and already has a tool that he used on Sub8 that is applicable to the tasks
* @DSsoto can write perception code quickly and me writing the mission code means that the tasks could be in a functional state sooner than later (we are beginning to get down to the wire)
* We have spoken about how this would be solved and it is clear that the perception and mission can be very cleanly split into separate tasks. Both can be independently managed and it will be clear which part is not working.

     I just think it would enable us to solve two challenges that are currently unsolved while efficiently utilizing our resources. Of course if these are all points you've heard before, I will differ to your experience.

whispercoros, 2016-11-15:
As for time to learn, I have cleared my schedule for the week so that I can solely focus on the project. I want to try to eliminate some large tasks before they come back to bite us.

tessbianchi, 2016-11-15:
You raise good points. 

However, we have three weeks until competition and the missions do not take long to write. There are multiple examples in the repo and many people in the lab know txros. So @DSsoto  can ask for help in the upcoming weeks if he needs. 

I agree that getting things working quickly is appealing, but balancing the work load between people in Hawaii is extremely important and I think taking this preventative measure early on is worth it. 

Thoughts?

whispercoros, 2016-11-16:
I think @DSsoto prefers to stick to just writing perception code. He has already worked on a lot of the backend and has written the perception for the hydrophone mission. In the case of that mission, the actual txros code was handled by @ironmig.

whispercoros, 2016-12-19:
Won without this because underwater visibility was garbage.

closed 2016-12-19 by whispercoros