uf-mil/NaviGator #134: Object misclassifications / Behaviors

ironmig, 2016-11-14:
Please leave as detailed descriptions of object misclassification you have noticed at the last few lakedays so we can get them resolved by competition. We should also discuss the desired behaviors of the object database, such as what to do when the type of an object changes.

I'll start

- Some totem markers present in the acoustic pinger challenge are not identified (remain unknown) preventing the gates from being identified
- Scan the Code is often identified as a totem

@maverick7170

ironmig, 2016-11-14:
Another thing to discuss: do we want to try any more complex classification methods? AFAIK, we classify only based on volume right now. Other options would be:

- Consider height vs width too
- Geometrical analysis (square, round, etc)
- Machine learning
- Ransac @whispercoros 
- Any combination of volume and the above methods

whispercoros, 2016-11-14:
We have the to-scale object models for each challenge element save one, which Lucas will get to me by the end of the week. If converting them to data that can be used by PCL is as easy as @tessbianchi says, it makes sense to pursue RANSAC because object classification is critical. I will be working on this with her throughout the week and hopefully it will be testable by the weekend.

tessbianchi, 2016-11-14:
I think this is how we should approach this problem:

1. Provide Ira with additional heuristics to add to the DB. For example, if something is isolated, then its not a buoy. Also, if it's not near a bunch of other buoys, and it's next to 3 similarly shaped objects, it's not a tower. He should be able to implement these by next test day.

2. Anthony is working on RANSAC this week.

3. I am going to try a machine learning approach and send Jake some data and he is going to try a machine learning approach. 

We have a double boat day so on Saturday we test all of these methods. We go with the best one on Sunday, and continue with that method until Hawaii. Anthony, and I will have no say on which one is 'the best', so that politics don't get involved. 

How does that sounds to everyone?

viron11111, 2016-11-14:
Sounds good to me!

whispercoros, 2016-11-15:
A competition to boost productivity, good thinking! Will there be trophies?

tessbianchi, 2016-11-15:
Loser has to buy everyone donuts

tessbianchi, 2016-11-16:
This is the email I just sent to Ira.

Hi Ira!

I have been thinking about how to approach object classification and here is what I think the algorithm should be. 

1. Throughout the run make sure check and see if each object is a buoy or a tower buoy, by volume. If it is, save it classify it, as you are doing now.

2. When you see something that is bigger than a buoy or a tower we will classify it as 'unknown'. Then, the explorer mission will go up to it, and make a service request to your code "classify_object <id>. The boat will then move around the object (60 or so degrees).

3. After this service call your classifier can gather points on the object for ~3 seconds and return what it thinks the object is. After that this object will stay in the database forever as this object.

----

I know this will take a little bit longer during our run. However, there are only 3 above water objects (other than buoys): Scan the code, Shooter, and Dock.

Also, we can optimize the explore mission, so that it looks for scan the code first (based on the fake markers). So that the other missions can complete and we don't have to backtrack.

I think this is the best method for what we want to do, because the lidar data that we are getting (from far away at least) is no good. 

-----

Let me know what you think.

 ~ Tess

whispercoros, 2016-11-16:
If I am understanding correctly that it is up to the explorer mission to select the object to take a closer look at at an arbitrary time, then I think this is a good behavior to implement.

tessbianchi, 2016-11-20:
It seems like most classifications work correctly if we drive up to objects rather than just start next to them. 

We just need every mission to implement their own mission specific queries to the database, and the first mission needs to be something that takes us far far away from all the objects.

maverick7170, 2016-11-20:
Sorry everyone, I haven't been checking the issues regularly. I was hoping today would go smoother with the classification so we could definitely choose a classification method. I agree with the Tess that classification seems fine when driving up to an objects and we account for reflections on sunny days. I only have 1 bag from today, but that bag works along with the bags from last week. So I'm getting good results in the bag data, but we still need trust this method will be reliable for competition. 

Was any progress made on RANSAC?

Also, any chance some bag data from today can be posted?

tessbianchi, 2016-11-20:
Not much progress on RANSAC. I think it might be too late to start anew, we need to work with the system we have.

Bags are uploading. They will be ready in T-35 minutes

maverick7170, 2016-11-20:
Ok, no problems with not having RANSAC. 
Thanks for the bags! I'll run these bags with the old bags and see what the issues are. Also, I'm going to stop by in lake in the morning before church!

tessbianchi, 2016-11-20:
Yay!!

See you then!

ironmig, 2016-11-23:
I think we're at a pretty good spot with this, so closing. Open a new issue with any more specific classification issues still remaining.

closed 2016-11-23 by ironmig