uf-mil/NaviGator #135: Write identify and dock perception / mission

ironmig, 2016-11-14:
By competition, we should have a mission written for the identify symbols and dock mission. It will likely use the same perception used for detect deliver (but running on the front camera). So far here is the plan:

- Generate a "fake" ogrid in the same region as the dock, with smooth lines representing the docking bays, so that obstacle avoidance can navigate inside the bays without collisions from the ogrid dilation.
- Determine the position and orientation of the three bays
- Move in front of each bay to identify the symbols using perception
- Navigate to the correct bay(s)

whispercoros, 2016-11-15:
I know there are issues with LiDAR and the dock in the simulator. My plate is full this week, but if someone else wants to look into it, I think the beam heights or resolution may be good things to tweak if they are easily modified in wamv.sdf.

ironmig, 2016-12-06:
Daniel, what is your current plan for this? If you can segment it into different parts, maybe me or someone else can work on it too. For example, if you define the interface for how to get the location of the bays, I could write the python mission. I think this is the only mission with nothing written in the repo for, and we only have a few more days before we start testing in Hawaii.

RustyBamboo, 2016-12-08:
I'm planning to have a toggleable topic that publishes the Ogrid for the dock. There will be a service to get center points of the bays and one unit normal vector.

ironmig, 2016-12-08:
Cool. I started working on the mission. Here's the service definition I'm working with:
```
---
geometry_msgs/Point[3] bays   #The positions in ENU frame of the center of the three bays 0=left, 1=center, 2=right
geometry_msgs/Vector3  normal #Vector or normal pointing away from plane of dock back
bool success                  #False if an error occured getting dock bays
string error                  #Descripion of error if success=false
```

whispercoros, 2016-12-18:
Many autonomous. Much success.

closed 2016-12-18 by whispercoros