uf-mil/NaviGator #136: Downward looking camera

viron11111, 2016-11-14:
Need TF for downward looking camera.  Camera will be mounted on 2x4 cross beam at a depth less than 1 meter.  Lucas has been notified.  Has anyone done the camera calibration for the camera?  The camera is in 121 on the file cabinet next to the window.

whispercoros, 2016-11-15:
@RustyBamboo is attempting to calibrate it at some point in the week, but @rleyva should be available on Thursday if he does not get it done by then.

RustyBamboo, 2016-11-15:
Udev wasn't working for me... Messed around with launch files. Coriander, camera1394 weren't working. Need help calibrating @rleyva

whispercoros, 2016-11-22:
Is this camera set up, or are we still waiting on it?

viron11111, 2016-11-22:
It's a Point Grey Firefly, USB 2.0.  Ralph has worked with this before.  Check with Lucas to see if the camera is physically mounted yet.  South Korea has already made progress towards underwater identification:
https://www.youtube.com/watch?v=aNcCalUy1_w

It's getting real!!!

RustyBamboo, 2016-11-22:
Solved in PR #154

closed 2016-11-22 by RustyBamboo