uf-mil/NaviGator #137: Acoustic based transit gotcha

viron11111, 2016-11-14:
The rules give an example of the pinger mission where the boat drives through one gate, circles a buoy, and then drives back through another gate (page 19, http://robotx.org/images/files/2016-MRC-Tasks-2016-11-03.pdf).  We'll have to be careful with how we define this mission since the pinger mission cannot be dependent on itself (ie, running a pinger mission immediately after a pinger mission, pinger mission dependent on pinger mission).  Recommend breaking the mission into three parts: pinger mission, circle mission, pinger mission.  Thoughts?

tessbianchi, 2016-11-15:
I think this should just be one big mission.

viron11111, 2016-11-22:
Roger.  Did we run this type of mission the weekend?

mattlangford, 2016-11-22:
We ran the pinger mission by itself on both sides of the gate, @ironmig says he set up code to circle the next totem we find.

ironmig, 2016-11-22:
I don't believe we will ever go through one gate and return through another. In quals, you go through the gate with the pinger, circle a totem 30 meters off, then go back through the same gate. In finals, you go through the correct gate to begin your run (instead of speed gate) and exit through the same gate to end.

viron11111, 2016-11-22:
Roger.

closed 2016-11-26 by whispercoros