uf-mil/NaviGator #139: Mission should not set other mission's inputs (Mission Planner should)

ironmig, 2016-11-15:
Considering the ambiguity in the rules of the use of mission outputs (scan the code colors, coral survey shapes, pinger gate) in determining mission inputs (shape to dock in, color to shoot at), I think we should abstract this decision making away from missions. Each mission should only set the parameters corresponding to the observed data (if any) of that mission. Then the mission planner should decide how to use that data to run other missions with different inputs. This way we have one file to edit for scripting all the missions together. Thoughts?

ironmig, 2016-11-15:
I still think it makes more sense for missions to be independent tasks that do not worry about what goes  after them. I think we should stick to the rule that one mission does not set another missions parameters, but only reads other mission's outputs, makes decisions about what to do, and sets its own outputs. 

For example:
- Scan the code runs, sets SCAN_THE_CODE_COLORS=[GREEN,BLUE,YELLOW]
- Coral Survey runs, sets CORAL_SURVEY_SHAPE=TRIANGLE
- Detect Deliver runs, reads scan the code and coral survey outputs, sets DETECT_DELIVER_COLOR=GREEN, DETECT_DELIVER_SHAPE=TRIANGLE

This way, if you want to see how a given mission decides what to do, you only have to look in that mission's file and no where else. I want to avoid the situation where we have to look through every mission to understand what's affecting one of them.

tessbianchi, 2016-11-15:
Ok I agree with that. 

This does not change anything on the mission planner level though.

whispercoros, 2016-11-15:
I agree with @ironmig on this one. We will almost certainly need to change which mission feeds into which in Hawaii and making this part of the YAML looks like the best way to go from my perspective. Is there anything particularly dangerous about specifying which param Detect and Deliver, for example, should pull it's colors or shapes from in the YAML?

tessbianchi, 2016-11-15:
Ok, I think I am misunderstanding what the initial intent of the issue was.

Let's discuss tomorrow.

whispercoros, 2016-11-22:
Can this be closed now that the mission planner has been updated and we have all settled on a solution?

closed 2016-11-22 by ironmig