uf-mil/NaviGator #151: Interactive Markers should not be relative position

tessbianchi, 2016-11-20:
After resetting on RVIZ the markers save their location and don't update. If there is a way to have the markers only update with absolute coordinates this would be best.

maverick7170, 2016-11-20:
This seems easy in theory but the documentation isn't really clear on this. My code sets the interactive markers to "enu" frame and then requests positions in a callback:

`int_marker.header.frame_id = "enu";`
....
`void roiCallBack( const visualization_msgs::InteractiveMarkerFeedbackConstPtr &feedback )
{
	auto newPose = feedback->pose;
        ....
}`

The pose reported from RVIZ isn't absolute though but relative moves in the enu frame. I'll read the docs and see what can be changed.

maverick7170, 2016-11-20:
I think I found part of the issue; I need to set the interactive marker server to the "enu" frame and leave then individual markers that are added to the server frame-less (instead of also setting the them to the "enu" frame). This seems to fix the Reset button issue in RVIZ but doesn't force everything to absolute coordinates.

tessbianchi, 2016-11-20:
Great thank you! I have bags for you

ironmig, 2016-11-24:
From skimming the docs, I think you're also supposed to check the ```event_type``` variable to be ```InteractiveMarkerFeedback::POSE_UPDATE```. It might be calling the callback when other things happen, so you have to make sure the type is pose update.

whispercoros, 2016-12-17:
Not happening

closed 2016-12-17 by whispercoros