uf-mil/NaviGator #163: Classification Smearing

ironmig, 2016-12-02:
Declare this outside of your callback, initialize in main function so that buffer is continuously updated with latest tf.
One potential improvement to the smearing could be synchronizing your tf.

```
tf2_ros::Buffer tfBuffer;
TransformListener tfListener(tfBuffer);
```

Then in your velodyne callback
```
//Duration is how long to block to wait for tf
  if (!tfBuffer.canTransform("enu", "velodyne",pcloud->header.stamp,  ros::Duration(1)))
  {
//Do something if tf can't happen
  }
  geometry_msgs::TransformStamped transform = tfBuffer.lookupTransform("enu", "velodyne", ,pcloud->header.stamp);
```
This ensures that your tf is synchronized to the time the pointcloud was produced.

maverick7170, 2016-12-02:
Thanks for looking into this! These changes didn't make a difference in my bag data but it's probably possible that it may matter at some point, so I'll leave the changes in.

tessbianchi, 2016-12-04:
It seems like close to nothing can be done for this problem, aside from going back in time and properly calibrating the lidar. 

I have write the image segmentation layer to object classification. This is a solution to the classification problems that we are having. Can I close this?

mgriessler, 2016-12-05:
I suggest we clone this issue to MIL's time machine project.

closed 2016-12-11 by tessbianchi