uf-mil/NaviGator #196: Switch CI to Jenkins

ironmig, 2017-08-22:
We should use the current CI system implemented on Subjugator, which uses our hosted Jenkins instance and docker containers for builds with github integration.

Overview of work needed:
* Create Dockerfile (use Subjugator's as a reference) to be used by CI for Navigator builds
* Create Jenkinsfile (use Subjugator's as a reference) to instruct the CI to test building, unit tests, and code policy
* Configure Jenkins through the web interface to trigger builds on pull request using the above mentioned container and Jenkinsfile
* Configure github webooks to trigger Jenkins and display check/x marks on build pass/fails

whispercoros, 2017-08-31:
Don't worry about adding any configurations to Jenkins or adding a webhook on Github. The daily organization scan should pick up the Jenkinsfile and automatically start building the NaviGator repository and there is a hook for the entire uf-mil Github organization that triggers Jenkins. If all goes well, adding the Docker image and Jenkinsfile should be all that you need to do.

ironmig, 2017-09-13:
Addressed in #219, now we just have to format our code to pass builds!

closed 2017-09-13 by ironmig