uf-mil/NaviGator #198: ICP to get rid of the Lidar Shmear

tessbianchi, 2017-08-23:
- Over time the lidar points would not show up in the same location over time. The current solution that I have in mind is using ICP to merge point clouds over time. 

- This ticket will also be used to update the status of the lidar.

mattlangford, 2017-09-02:
Do you think that the conditions we had in Hawaii last year would mess up ICP? Particularly being far away from the land where everything is moving around you (in high currents and high winds). It seems like a good alternative to ICP might be calibrating a better INS->Velodyne transform. Thoughts?

mgriessler, 2017-09-02:
We have to recalibrate the Velodyne anyway, we might as well do it as well as possible.