uf-mil/NaviGator #204: Basic SIM integration test

ironmig, 2017-08-31:
We should have a test to verify that the simulator works in a basic sense (without crashing, plugins missing, etc)

A nice integration test to do this is as follows:
* In a rostest, launch the gazebo simulator
* Simply verify that basic topics are received: odom, absodom, camera, lidar points
* Do some basic verification of these topic's data (odom starts at 0, camera images are non-empty, lidar points are non-empty)

The following stretch goals would test a lot of the functionality of the SIM and some of our other software stacks:
* Execute move command(s) in SIM (forward, backward, left, right, yaw), verify that when the move is complete Navigator is now at the correct position/orientation

jmmarotta, 2017-09-28:
All goals except for the stretch goals are complete.

ironmig, 2017-11-16:
#241

closed 2017-11-16 by ironmig