uf-mil/NaviGator #206: Basic Alarms test

ironmig, 2017-08-31:
Though ros_alarms has unit testing, we should add some simple tests to verify our alarm handlers.

* broadcast a series of odom messages which are discontinuous, verify odom-kill raises
* broadcast the network hearbeat than stop, verify network-loss is raised
* raise hw-kill to verify the meta alarm is processed and kill raises

RustyBamboo, 2017-11-01:
Please also add thruster alarm test, which can be seen in #271

ironmig, 2018-01-19:
Any work on this? I'd like to give someone else the task if you don't think you'll get a chance to finish it

ngynkvn, 2018-01-19:
@ironmig I haven't really had time to do this, you can assign it to someone else!

ironmig, 2018-01-19:
Cool, feel free to check out some of the other issues if you still want to be involved in MIL.