uf-mil/NaviGator #226: Format python files

ironmig, 2017-09-14:
Our CI software checks [for python compliance with pep8](https://github.com/uf-mil/NaviGator/blob/kinetic-upgrade/Jenkinsfile), a formating standard for python. You can check if the python files are properly formating by running ```python2.7 -m flake8 --ignore E731 --max-line-length=120 --exclude=__init__.py <NAVIGATOR SOURCE FOLDER>```. For our builds to pass, someone needs to make all of our python files compliant (on kinetic-upgrade branch).

* A lot of the issues can be fixed by runing autopep8, an auto formatter
* There will likely be some remaining issues to fix, such as unused imports. For these you will need to manually edit the files until no errors show up when running the above command

yikaimao, 2017-09-24:
Are we currently working on circle_tower.py? I'm almost finished and 80% of errors left are from this file...

yikaimao, 2017-09-24:
also for this file:
NaviGator-kinetic-upgrade/deprecated/utils/templates/template.py

Since the name is "template", I didn't do any change to the unused importations, is that OK?

ironmig, 2017-09-25:
So we need your PR to pass the python format check on CI. If you click on the [X mark on your PR](https://ci.mil.ufl.edu/jenkins/blue/organizations/jenkins/uf-mil%2FNaviGator/detail/PR-237/1/pipeline) you can see the remaining issues. If think it isn't worth it to format a file, you can tell flake8 to ignore it by putting ```# flake8: noqa``` at the top of the file or on individual lines to skip. Also, you can ignore anything in the deprecated folder (including template.py), but you will need to tell the CI to not format check this folder by adding ```--exclude=./deprecated``` to line 27 of [Jenkinsfile](https://github.com/uf-mil/NaviGator/blob/kinetic-upgrade/Jenkinsfile).

ironmig, 2017-09-25:
You should add to the Jenkinsfile to exclude files in our submodules, because we have no control of them. ```--exclude=./gnc/navigator_path_planner/lqRRT```

closed 2017-10-29 by ironmig