uf-mil/NaviGator #238: Restore simulation of wrench/thruster forces

ironmig, 2017-09-25:
As described in the sprint plan, we want to have the option at runtime of the simulator to have gazebo simulate our current wrench or even the forces on each individual thruster. There is some code already present to do this (maybe), so one just has to get it running again.

mattlangford, 2017-09-26:
This didn't work in sim last year right? Like the boat would just go wild since the controller is tuned for real life boat.

ironmig, 2017-09-26:
Yeah, that's why we switched to the carrot follower, which will continue to work as the default. We just want to be able to test changes to any code lower in the stack in SIM. Perhaps we'll tune independently for the SIM or just accept instability when we're testing these things

ironmig, 2018-01-29:
We're going to use the 2D sim #293 instead for testing this.

closed 2018-01-29 by ironmig