uf-mil/NaviGator #252: Allow multiple commands in move command

ironmig, 2017-10-22:
Right now we have a [move script](https://github.com/uf-mil/NaviGator/blob/master/mission_control/navigator_missions/nodes/move_command) which allows input like
```nmove forward 2``` -> Moves forward 2 meters

[On Subjugator](https://github.com/uf-mil/SubjuGator/blob/master/command/sub8_missions/nodes/move_command), we have a similar system but it allows multiple commands at once, expected one of the next
```smove forward 2 yaw_right 30deg``` -> move forward 2 meters then yaw right 30 degrees

Add this functionality to NaviGator's system

ironmig, 2017-10-22:
Note: This will change _slightly_ with the new mission system I'm working on, so I recommend you work off that. It can be found on my [kinetic-upgrade-missions branch](https://github.com/ironmig/Navigator/blob/kinetic-upgrade-missions/mission_control/navigator_missions/navigator_missions/move.py)

ironmig, 2017-10-26:
Nick

ironmig, 2017-11-05:
Some hints:

You'll want to test this in the simulator. To run the simulator (it will slow down your computer)
```roslaunch navigator_launch simulation.launch gui:=false```

To run the command
```navc forward 2 left 1``` for example

thefata, 2017-11-05:
I have the script completed, however I have been unable to run the simulator using the command you provided.

RustyBamboo, 2017-11-05:
Check out [16.04 upgrade](https://github.com/uf-mil/NaviGator/wiki/16.04-Upgrade) and [running simulator wiki](https://github.com/uf-mil/NaviGator/wiki/16.04-Upgrade)

ironmig, 2017-11-16:
#274

closed 2017-11-16 by ironmig