uf-mil/NaviGator #254: Address Battery Paradox

ironmig, 2017-10-22:
We discovered an interesting bug last test day. Our kill system uses the voltage readings from our motor controllers to kill NaviGator when the battery gets too low, and warn us before this. But when a kill is triggered, the motor controllers stop getting their main power. However, they still send feedback over USB, but report a very low / sporatic voltage. This means that once we are killed, the measured voltage will drop (inaccurately) and not allow us to unkill. 

A few options to address this:
* When we are killed, ignore the voltage readings from the motor controllers
* If the motor controllers report a undervoltage fault or other indicator that they are not powered, ignore the readings
* Lobby to get a real voltage sensor on the robot

mgriessler, 2017-10-23:
The new battery management system being worked on will include voltage sensing from each battery. So you will also know if both batteries are powered. It will be beta tested no later than the beginning of spring semester and as long as a respin isn't necessary, it'll be installed by the end of January 2018.

However, I also suggest bullet B.

ironmig, 2017-10-26:
Reminder! Re-enable battery voltage meta alarm when this is done

ironmig, 2017-11-02:
Voltage readings are now ignored if we are hardware killed.

closed 2017-11-02 by ironmig