uf-mil/NaviGator #266: Thurster Visualization Tool

ironmig, 2017-10-27:
On Subjugator, we have a [thrust visualization tool](https://github.com/uf-mil/SubjuGator/blob/master/utils/sub8_diagnostics/scripts/thruster_visualizer.py), which publishes rviz markers displaying an arrow coming out of each thruster scaled to it's current commanded thrust. Make a similar tool for NaviGator. Subscribe to the thrust commands, load the thrust positions from the params, and publish markers to an rviz markers topic.

ironmig, 2017-10-29:
@andrew-helms I think this would be a good one for you

andrew-helms, 2017-10-29:
I think so too, it sounds really interesting.

ironmig, 2017-11-04:
Some hints:

* To set the thruster position [ROS param](http://wiki.ros.org/Parameter%20Server), launch the GNC stack:```roslaunch navigator_launch gnc.launch```

* You can see the parameters now in the terminal
```
$ rosparam get /thrust_mapper/
effort_limit: 400
effort_ratio: 1.35
thruster_BL_cog: [-1.9304, 1.016]
thruster_BL_theta: 0.785398
thruster_BR_cog: [-1.9304, -1.016]
thruster_BR_theta: -0.785398
thruster_FL_cog: [1.5748, 0.6096]
thruster_FL_theta: -0.785398
thruster_FR_cog: [1.5748, -0.6096]
thruster_FR_theta: 0.785398
```

* To simulate a thruster command to FrontLeft motor
```
rostopic pub /FL_motor/cmd roboteq_msgs/Command "mode: 0
setpoint: 100.0" 
```

ironmig, 2018-01-19:
Any work on this? I'd like to give someone else the task if you don't think you'll get a chance to finish it

andrew-helms, 2018-01-19:
No, sorry. I was still trying to get used to python and the MIL framework when my old laptop died. I just got the new one in so I can start working on it again. If it's possible, it would be nice if I could work with someone else on it to learn ROS and python.

ironmig, 2018-01-19:
Sure! Any of the veteran software people can answer questions over slack or in MIL if you get stuck. We'll have our first meeting of the semester next week and if there someone else who wants to work on it you can partner up.