uf-mil/NaviGator #269: Find all python files (even ones that don't end in .py)

ironmig, 2017-10-29:
For format checking and other purposes, we need a script that will print a list of all python files in a directory (recursively down subdirectories too). Right now we just run find piped into grep for .py, but some python files are scripts that don't end in .py. 

Print all files that EITHER
* end in .py
* have ```#!/usr/bin/env python``` as the first line

brooksturtle, 2017-12-03:
[pythonProgramList.zip](https://github.com/uf-mil/NaviGator/files/1525286/pythonProgramList.zip)

brooksturtle, 2017-12-03:
This should work, everything is in the readme

brooksturtle, 2017-12-03:
Combined a bash and python file, let me know if it works for you

ironmig, 2017-12-04:
Cool....  a little over complicated though. I think this could just be one python script [os.walk](https://docs.python.org/2/library/os.html#os.walk). Please PR it to mil_tools when this is done (it should be generic to run in any directory).

ironmig, 2018-03-30:
Merged

closed 2018-03-30 by ironmig