uf-mil/NaviGator #278: Move command SIM Test

ironmig, 2017-11-16:
Test certain moves in SIM, verifying that we get to the goal waypoint and follow something close to the right trajectory to get there. This will verify the move command / mission system (please test with #274 and uf-mil/mil_common#87), the trajectory generator, and whole lot more.

ironmig, 2017-11-16:
I'm thinking @thefata on this one

thefata, 2018-02-06:
After pull the most recent updates from GitHub, nmove is the proper command and when run, it waits for the action server, which I located and launched. However, when launching the action server (task runner), a warning about TCP/IP failure returns and the server instantly quits after launch. After further inspection of the log files returned, I found the source of the problem to be the task_runner.py in mil_common, which is throwing an import error.

ironmig, 2018-02-06:
Huh I'll look into it. This might be something I fixed in #294

ironmig, 2018-03-14:
@thefata the 2D sim is merged and ready to go, which should make it much easier to write this test. [Here is some usage info](https://github.com/uf-mil/NaviGator/wiki/How-To-Run-The-Simulator). Basically, this test will just run a series of move commands and verify that we reached the goal by checking /odom. The test must be written using the rostest framework. You can use [this one](https://github.com/uf-mil/NaviGator/blob/master/hardware_drivers/navigator_kill_board/test/kill_board_test.py) as an example. You got this!