uf-mil/NaviGator #281: Scan the code firmware: repeat pattern for n seconds

ironmig, 2017-11-16:
As we discussed in the meeting, update the arduino code for scan the code to loop in the following way
* Randomly select 3 colors (out of Red, Green, Blue, Yellow) to flash
* Flash that pattern (1 second per color, one second off) for n seconds
* repeat

This will allow us to test multiple patterns without reprogramming the arduino, but still have a pattern stay on long enough to test.

yikaimao, 2018-01-19:
the LED panel is now updated with new code.
some potential problems:
1. delay is not exactly 1 sec long, feels a little bit longer(not sure why but I have to use delay(2000) instead of delay(1000) to get about 1s delay)
2. yellow is too bright, looks like white(now using 4-bit color mode, change RGB to (15, 10,0) is a little bit better)
3. old code draw each pixel one by one, new code uses the matrix.fillScreen() function. (maybe slower, maybe faster, needs to be tested by the new camera)

ironmig, 2018-01-19:
Cool it does look better! However I see a significant delay between displaying colors in the same sequence. It should go RED GREEN BLUE OFF with no off in between colors. I'm also not seeing the change described in the issue, which is to display the same sequence several times before switching.
[image]

yikaimao, 2018-01-19:
My bad, i read the rules wrong. I thought there should be 1s off between each color, I’ll remove the delay! 
How many times you want to repeat the same pattern? 3 maybe?

yikaimao, 2018-01-19:
Also can you distinguish between yellow and green from the camera image? I can display characters on the panel so if you want more functions just let me know

ironmig, 2018-01-22:
Yes yellow and green are distinct. I'd say repeat a pattern 5 times.

ironmig, 2018-01-25:
Thanks @yikaimao! this is done.

closed 2018-01-25 by ironmig