uf-mil/NaviGator #282: Bug: Goal/seed out of bounds of occupancy grid

rknlhrqy, 2017-11-19:
In function select_exploration() in file lqrrt_node.py, you have:
occ_img_dial[seed[1], seed[0]]
occ_img_dial[goal[1], goal[0]]

occ_img_dial is an array with each element showing the distance in meter from the starting point [0, 0].
seed and goal are two tuples in cell. 
It can cause "out of range" in occ_img_dial.
For example, if I set the goal to be at (700,700) meter. Then goal will be (3200,3200) cell. And occ_img_dial[3200, 3200] is out of range. 

This is my first time to open issue in this code. Please let me know if it is appropriate. 
Thanks.
--Kening

jnez71, 2017-11-19:
Lines 573-579 from lqrrt_node.py,
```
# Make sure seed and goal are physically meaningful
try:
    occ_img_dial[seed[1], seed[0]]
    occ_img_dial[goal[1], goal[0]]
except IndexError:
    print("Goal and/or seed out of bounds of occupancy grid!")
    return(0, escape.gen_ss(self.next_seed, self.goal, 1), np.copy(self.goal), False)
```
verify that both the seed position (origin of path planning) and the goal (user-specified waypoint position) exist inside the look-up bounds of the occupancy grid. Note that occ_img_dial is just a dilated version of occ_img which is just the occupancy grid interpreted as an image. (I realize now that "dial" was a bad suffix since "dialated" is a misspelling of "dilated" lol...) Regardless it should be the same size as occ_img (dilation doesn't affect the shape of an image, only its contents).

If we request to go to (700, 700) when the occupancy grid is only say, 100 by 100, we are trying to move out of bounds, and so the try-catch from those lines of code results in a warning and the termination of the select_exploration function (which analyzes the occupancy grid to choose heuristics for the current move) by returning some benign defaults.

From there, the boat is SUPPOSED to try to drive to the closest feasible point towards the given infeasible goal, but instead it freaks out. This is a bug! I found the issue. I think it was caused by a numpy update that forbid list multiplication by numpy floats... or something like that. I will make a Pull Request to fix this.

Anyway, in general if you are trying to drive outside the occupancy grid (why?), set blind=True in your move action.

jnez71, 2017-11-19:
I have made the [PR](https://github.com/uf-mil/NaviGator/pull/283/files), which definitely fixes *some* issue but I don't know if it fixes the specific issue you're pointing out (which just sounds like you can't get the boat to drive out of occupancy-grid bounds; use blind=True for that).

rknlhrqy, 2017-11-21:
Jason,
Thank you for the quick response. After reading your feedback and studying it again, I realize that I opened a wrong issue. The code in question works fine. I now close the issue.

Thank you.
--Kening

closed 2017-11-21 by rknlhrqy