uf-mil/NaviGator #290: Display NaviGator over satilite imagery

ironmig, 2018-01-19:
NaviGator has a GPS and absolute (earth fixed) orientation sensors, so it is possible to display NaviGator's current position (and potentially other detected objects) over satellite imagery from open street maps or w/e. Not only would it look really cool, but it would help understand where we are at test days / competition when we can't see NaviGator easily!

* Google if there is some  easy way to do this in ROS already with an rviz / rqt plugin
* Otherwise, research some options to roll this out ourselves (web app is okay)

joshrudi, 2018-02-23:
Just an update on this.  It appears it IS possible to do this in ROS via "rviz".  I've been playing around with getting the testing environment running in a Linux VM on my Mac, with many roadblocks.  Will update when I make progress.

ironmig, 2018-02-26:
Cool! You can always test on the "shuttle" computer in the Lab to test. Dual booting is also an option, you can find resources for this online

ironmig, 2018-04-09:
@RustyBamboo found a way to do this

joshrudi, 2018-04-09:
Great!  Kudos @RustyBamboo

joshrudi, 2018-04-09:
Just for future reference, here's a link for the built-in ROS method.  http://wiki.ros.org/gpsd_viewer#Usage

ironmig, 2018-11-13:
@thedpws updates?

thedpws, 2018-11-19:
It's worked for all the bags I've tested, so I'm submitting it for PR! https://github.com/uf-mil/NaviGator/pull/434

PR #434 (2018-11-23): SATELLITE: Display NaviGator on satellite imagery
closes #290 

To launch: `roslaunch rviz_satellite launch.launch`
This package (rviz_satellite) subscribes to `/odom` and `/lla` to display NaviGator on satellite imagery in rviz.

The launch file will launch rviz with the `satellite.rviz` configuration, as well as the node for converting and publishing to `/lla`. 

`satellite.rviz` is based on `navigator.rviz`, but has **all elements scaled up by a factor of 20.**
closed 2018-11-23 by ironmig