uf-mil/NaviGator #291: Anti-LED scanning computer vision node

ironmig, 2018-01-19:
The scan the code challenge requires us to determine the color of an LED panel. One of the difficulties of this is that the LED panel has a refresh rate which shows scanning lines in images instead of the solid colors our eyes see. Besides working on this from the camera side, it should be possible to process an image to reduce this affect. Here's my idea:
* have NaviGator stay still while looking at the LED panel
* capture several images
* use an algorithm to smear these images together in a given region. If a pixel is near black, use the non-black pixel from a nearby frame. This should produce a more solid color in this part of the image with the scanning lines
 
[image]

ironmig, 2018-04-03:
Progress update @gdutk007 ?

gdutk007, 2018-04-04:
unfortunately haven't made much progress, had to stop working on the assignment couple weeks into the semester. If anyone else is interested in working on it you can feel free to reassign

ironmig, 2018-04-04:
Seems like a histogram approach may be better, clossing

closed 2018-04-04 by ironmig