uf-mil/NaviGator #292: Computer Vision based Buoy detector

ironmig, 2018-01-19:
We have a pretty serious problem on NaviGator which affects our obstacle avoidance ability: the course if filled with black buoys which we are penalized for colliding with. If they our detected, we can easily plan a trajectory around them. Unfortunately, our object detection system is entirely LIDAR based, which does not get good returns of black objects. We often only get 1 return off a buoy, which is filtered out as it is difficult to distinguish this from normal erroneous water reflections at the same height. It seems pretty clear that we need to use computer vision here. This is my idea:

* for any potential buoys (few returns at a low height), use the 3D points to find a region of interest in the nearest camera to check if it is a buoy or just water
* use computer vision (traditional or machine learned) to return if this point should be kept
* Neural network ???
* Profit

ironmig, 2018-01-29:
Another interesting idea is using the camera we found that seems to be responsive to UV spectrum. Perhaps the buoys would look distinct against the background with this camera.

RustyBamboo, 2018-01-29:
Is the camera USB? Think we can have it mounted/plugged in for Sunday to see how it looks?

ironmig, 2018-01-29:
Yeah its plug and play USB. I'll ask the greaseys

ironmig, 2018-12-04:
replaced by #360

closed 2018-12-04 by ironmig