uf-mil/NaviGator #293: Simple 2D GNC simulator

ironmig, 2018-01-19:
Our current simulator, based on gazebo, is essentially a 3D game engine which is incredibly computational expensive. This makes it difficult to use for quick tests, automated integration tests, or at all on some computers. We really don't need all of the features gazebo offers for some purposes. 

Find and configure OR create a 2D simulator which will just allow us to test path planning, the controller, thruster mapper, etc. NaviGator can just be in a 2D with no objects. When the thrusters output forces, NaviGator moves against friction! It doesn't have to be a perfect physics simulation, as it will mostly be for debugging.

ironmig, 2018-01-26:
Let me set some goals:

- [ ] pull all constants (mass, inertia, buoyancy) from ros params
- [ ] Publish to /odom with current state (you can ignore covariance)
- [ ] Subscribe to /wrench, apply force/torque from wrench to NaviGator center of mass
- [ ] Optionally, subscribe to thruster thrusts and apply forces as thrusters instead of wrench. Thruster positions/force mappings are already set in launch files
- [ ] Set mass/inertia/etc constants to something close to real values using bag data + math + smart

RustyBamboo, 2018-02-01:
http://wiki.ros.org/stdr_simulator/

Grymestone, 2018-02-14:
- [x] pull all constants (mass, inertia, buoyancy) from ros params
- [ x] Publish to /odom with current state (you can ignore covariance)
- [ x] Subscribe to /wrench, apply force/torque from wrench to NaviGator center of mass
- [ ] Optionally, subscribe to thruster thrusts and apply forces as thrusters instead of wrench. Thruster     positions/force mappings are already set in launch files
- [ ] <Progress> Set mass/inertia/etc constants to something close to real values using bag data + math + smart
- [ ] Unit test to verify functionality (publish a wrench, see that NaviGator moves as it should)
- [ ] Publish to /absodom
- [ ] Fix Launchfile to spin everything up (rviz, xbox controller, alarms, shore_launch, etc) so it is ready to sim through a single command.

ironmig, 2018-02-23:
Going to create individual issues for the remaining tasks as the core SIM is now merged.

closed 2018-02-23 by ironmig